TMRh20/nrf_to_nrf #1: Static Payloads not working

TMRh20, 2022-12-07:
Logging this issue for tracking purposes and so users are aware of the issue

**Sending static payloads**: It seems payloads are sending OK, but the ACK results in a CRC error on the NRF52 side

**Receving static payloads**: Payloads are received by nrf52 but an ack is not received on the nrf24l01 side

Still working on this issue but not sure what to try at this point.

TMRh20, 2023-05-06:
I've been working on this issue a bit, and came up with an interesting solution for sending static payloads from NRF52. It seems that if I simply disable CRC checking before receiving an ACK, it works fine. Not sure if this is the way to do things, if they really disable CRC checking on ACKs, but I guess it would make sense that if ACK-Payloads aren't enabled (needs DPL enabled), there would be no need to do CRC checking on ACK payloads really. 

Will keep working on the issue and post my fix soon if I can't get reception of static payloads working properly on NRF52 as well.

TMRh20, 2023-05-07:
## Current Status:
**Sending static payloads:** Working
**Receiving static payloads** Ack is not received on NRF24

2bndy5, 2023-05-07:
Not sure if this helps... The nRF5 SDK doesn't seem to use the `EVENTS_CRCERROR`, but they do check the `CRCSTATUS` when entering the callback for receiving an ACK during TX mode.

```c
static void on_radio_disabled_tx_wait_for_ack()
{
    // This marks the completion of a TX_RX sequence (TX with ACK)

    // Make sure the timer will not deactivate the radio if a packet is received
    NRF_PPI->CHENCLR = (1 << NRF_ESB_PPI_TIMER_START) |
                       (1 << NRF_ESB_PPI_RX_TIMEOUT)  |
                       (1 << NRF_ESB_PPI_TIMER_STOP);

    // If the radio has received a packet and the CRC status is OK
    if (NRF_RADIO->EVENTS_END && NRF_RADIO->CRCSTATUS != 0)
    {
        NRF_ESB_SYS_TIMER->TASKS_SHUTDOWN = 1;
        NRF_PPI->CHENCLR = (1 << NRF_ESB_PPI_TX_START);
        m_interrupt_flags |= NRF_ESB_INT_TX_SUCCESS_MSK;
        m_last_tx_attempts = m_config_local.retransmit_count - m_retransmits_remaining + 1;
```

> if ACK-Payloads aren't enabled (needs DPL enabled), there would be no need to do CRC checking on ACK payloads

IDK, the datasheet does state that the CRC is "calculated over the address, Packet Control Field and Payload." So, even if there is no payload in the packet, then the Packet Control Field's PID could influence the CRC checksum, although this doesn't seem too robust.

TMRh20, 2023-05-07:
Hmm, so it seems they do check CRC on RX of ACK. Its odd that both devices are seemingly rejecting each others CRC. Pretty sure that's what's happening on the nrf24 end of things still.

2bndy5, 2023-05-08:
I guess it depends on what is considered an empty payload in the ACK packet. If the nRF24 uses a null byte and the nRF5x omits a value all together (or vice-versa), then the CRC would be inconsistent.

BTW, regarding
https://github.com/TMRh20/nrf_to_nrf/blob/adc651f183dc87bffeff0aa2e193f6a14ddc52bb/src/nrf_to_nrf.cpp#L666-L678
The values are correct but not the comments.

TMRh20, 2023-05-08:
Yeah, thats what copy/paste gets you lol. Will have to remember to fix that.

Either way, it seems strange to me that it works fine using dynamic payloads, the initial values pass CRC checking with DPL disabled, its just the static ACKS that fail CRC on _both_ sides.

2bndy5, 2023-05-08:
It is strange. I'm getting hung up on the claim that payloads cannot be 0 bytes as stated in multiple places (NRF24L01 data sheet and nRF5 SDK's docs), but this claim may or may not be applicable to intentionally empty packets. Looking at the SDK, they only update the packet buffer's payload_length value. They also arbitrarily copy payload data to the buffer, but that essentially uses `memcpy(dst, src, 0)` (effectively a non-op).

<details><summary>How the SDK updates payload info</summary>

```c
static void update_rf_payload_format_esb_dpl(uint32_t payload_length)
{
#if (NRF_ESB_MAX_PAYLOAD_LENGTH <= 32)
    // Using 6 bits for length
    NRF_RADIO->PCNF0 = (0 << RADIO_PCNF0_S0LEN_Pos) |
                       (6 << RADIO_PCNF0_LFLEN_Pos) |
                       (3 << RADIO_PCNF0_S1LEN_Pos) ;
#else
    // Using 8 bits for length
    NRF_RADIO->PCNF0 = (0 << RADIO_PCNF0_S0LEN_Pos) |
                       (8 << RADIO_PCNF0_LFLEN_Pos) |
                       (3 << RADIO_PCNF0_S1LEN_Pos) ;
#endif
    NRF_RADIO->PCNF1 = (RADIO_PCNF1_WHITEEN_Disabled    << RADIO_PCNF1_WHITEEN_Pos) |
                       (RADIO_PCNF1_ENDIAN_Big          << RADIO_PCNF1_ENDIAN_Pos)  |
                       ((m_esb_addr.addr_length - 1)    << RADIO_PCNF1_BALEN_Pos)   |
                       (0                               << RADIO_PCNF1_STATLEN_Pos) |
                       (NRF_ESB_MAX_PAYLOAD_LENGTH      << RADIO_PCNF1_MAXLEN_Pos);
}

static void update_rf_payload_format_esb(uint32_t payload_length)
{
    NRF_RADIO->PCNF0 = (1 << RADIO_PCNF0_S0LEN_Pos) |
                       (0 << RADIO_PCNF0_LFLEN_Pos) |
                       (1 << RADIO_PCNF0_S1LEN_Pos);

    NRF_RADIO->PCNF1 = (RADIO_PCNF1_WHITEEN_Disabled    << RADIO_PCNF1_WHITEEN_Pos) |
                       (RADIO_PCNF1_ENDIAN_Big          << RADIO_PCNF1_ENDIAN_Pos)  |
                       ((m_esb_addr.addr_length - 1)    << RADIO_PCNF1_BALEN_Pos)   |
                       (payload_length                  << RADIO_PCNF1_STATLEN_Pos) |
                       (payload_length                  << RADIO_PCNF1_MAXLEN_Pos);
}
``` 

</details>

TMRh20, 2023-05-08:
If I disable the 'fix' and treat it as though the acks are 4-byte static payloads like the main payloads, then I get the following output, printing the radio buffer and received CRC from the NRF52 register RXCRC. 

This is the 5 retried acks coming from the NRF24 to the NRF52:

```
03:53:13.007 -> CE:0:E:CA:34:32: CRC: 8394
03:53:13.007 -> CF:0:2E:88:29:77: CRC: 9170
03:53:13.007 -> CC:0:4E:4E:13:A4: CRC: AC33
03:53:13.007 -> CD:0:6E:C:3E:42: CRC: CC1E
03:53:13.007 -> CE:0:E:CA:2C:3E: CRC: 42DC
03:53:13.007 -> Transmission failed or timed out
03:53:14.032 -> CF:0:2E:88:35:2F: CRC: 8E15
03:53:14.032 -> CC:0:4E:4E:31:A: CRC: 216B
03:53:14.032 -> CD:0:6E:C:5:8F: CRC: D032
03:53:14.032 -> CE:0:E:CA:30:1B: CRC: 63FD
03:53:14.032 -> CF:0:2E:88:20:B: CRC: 4CEA
```

2bndy5, 2023-05-08:
Interesting, the first 4 bytes of the buffer data seem to be consistent per PID. Is the CRC checksum similarly consistent when expecting a 0-length payload?

You're doing well to summon my curiosity now, but I'm a bit preoccupied ATM to fire up my own hardware. Hopefully, this week I'll start playing around with this.

2bndy5, 2023-05-08:
Oh wait, that is the originally intended Packet Control Field & CRC checksum. 🤦🏼

TMRh20, 2023-05-09:
I opened an issue with Nordic to see if we can get a bit of help 
https://devzone.nordicsemi.com/f/nordic-q-a/99452/nrf52840-to-nrf24l01-communication-with-static-payloads

TMRh20, 2023-05-09:
Well that was easy. LOL.

For the record, need to set payload length to 0 before incoming or outgoing ack while using static payloads. 
Commit incoming.

2bndy5, 2023-05-09:
Yeah, I saw that 🎉 . It is a special kind of awesome when the fix turns out to be something as easy as a few lines and so much time was spent investigating. 😜

commit e9a7798e (2023-05-09): Fix #1 issue

Closes #1

Set Payload size to 0 before sending or receiving ACK if DPL is enabled.
closed 2023-05-09 by TMRh20