TMRh20/nrf_to_nrf #4: NRF24L01+ configuration does not respond properly

Matorra, 2023-02-12:
I am using the library RF24 version=1.4.5 author=TMRh20

Although the communication seems to work, when I want to change some parameters such as the emission power or the frequency of the communication I see that when printing the configuration data with the instruction radio.printPrettyDetails(); I don't see that these values change on the monitor serial. For example, if I change the SPI frequency, if the parameter is modified in the data printout, but not the power, or the channel, or the number of repetitions and the waiting time between them.

 the setup is:

RF24 radio( pinCE , pinCSN,4000000 );
...
void setup()
{
   Serial.begin(115200);   // Inicializamos  el puerto serie  
   pinMode(Buzer, OUTPUT);
   pinMode(PITO, OUTPUT);
   pinMode(LED_BUILTIN,OUTPUT);
   radio.begin();
....
   radio.setChannel(124);
   radio.setRetries(10,15);
   radio.setPALevel(RF24_PA_LOW);// RF24_PA_HIGH
   radio.setDataRate(RF24_250KBPS);
   radio.setAutoAck(true);
   delay( 500);
  if (!radio.begin()) {Serial.println("radio hardware is not responding!!"); }else {Serial.println("radio OK"); }
  radio.openReadingPipe(1,0xAFBFCF0F01LL);
  radio.openReadingPipe(2,0xAFBFCF0F02LL);
  radio.openReadingPipe(0,0xAFBFCF0F00LL);
  radio.openWritingPipe(0xAFBFCF0F05LL);
  radio.startListening();
  radio.printPrettyDetails();
}

The result on the serial monitor:
(I indicate in red the parameters that I have changed without adequate response,) However, the communication occurs and although it is not 100% stable, it seems to be correct. I am using two different boards an ESP32 and an ESP32 C3 with the same result.

radio OK
SPI Frequency           = 4 Mhz
Channel                 = 76 (~ 2476 MHz)
Model                   = nRF24L01+
RF Data Rate            = **1 MBPS**
RF Power Amplifier      = **PA_MAX**
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = **1500 microseconds**
Auto Retry Attempts     = 15 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 15
Multicast               = Disabled
Custom ACK Payload      = Disabled
Dynamic Payloads        = Disabled
Auto Acknowledgment     = Enabled
Primary Mode            = RX
TX address              = 0xafbfcf0f05
pipe 0 (closed) bound   = 0xafbfcf0f05
pipe 1 ( open ) bound   = 0xafbfcf0f01
pipe 2 ( open ) bound   = 0x02
pipe 3 ( open ) bound   = 0x00
pipe 4 (closed) bound   = 0xc5
pipe 5 (closed) bound   = 0xc6

I would appreciate any help on the subject.

Matorra, 2023-02-12:
I have already located the problem, I understand that I have use two times  radio.begin()  and the second time it deletes the initial data.

TMRh20, 2023-02-13:
Thanks for posting the resolution!

closed 2023-02-13 by TMRh20