TMRh20/nrf_to_nrf #5: startWrite() function not working properly

TMRh20, 2023-04-22:
This function will probably never work properly until/unless we get interrupts working for these devices. Opening an issue for tracking purposes.

2bndy5, 2023-04-22:
Maybe `txStandBy()` could be put to better use here. Although, that idea might imply different behavior from `RF24::txStandBy()`.

TMRh20, 2023-04-23:
How? You assume I know what I'm doing or something. :p
Without interrupts we can't really do anything in the background, so I don't think we can get the functionality working akin to RF24.

2bndy5, 2023-04-23:
yeah it wouldn't work like in RF24.

```c
startWrite();
txStandBy();
```
Indeed, we'd need interrupt support to implement like RF24 does, but I was thinking in a less-blocking way than `write()`.

TMRh20, 2023-04-23:
Hmm, that might be worth looking into. 

I really should get at least one more of these devices, they are pretty nice and I could troubleshoot more effectively.

2bndy5, 2023-04-23:
I picked up the XIAO expansion board and a ST-Link V2 so I could do even more.

I'd recommend a Microbit because it has a builtin debugging chip, but it has little Arduino support. Microbits are better for using MicroPython or Zephyr Project.

commit b541285b (2023-05-28): startWrite function, adjustments

- Remove unnecessary calls to TASKS_START (using shortcut)
- Set lastTxResult in write and startWrite
- Adjust startWrite to just send a packet and not wait for completion
- Adjust txStandBy to wait for TX completion and put radio into disabled state
- Update docs
- Closes #5
closed 2023-05-28 by TMRh20