TMRh20/nrf_to_nrf #7: Using the CCM - AES CCM mode encryption

TMRh20, 2023-05-14:
Just wondering if this is something that should be included with the library.
I've mocked up some simple code to make use of the feature, but I'm not all that familiar with managing encryption and authentication.

In any case, it seems to work well, and allows a pretty seamless interaction:
1. Using static payloads, the payload size needs to be set 4 bytes larger than the payloads being sent/received
2. Using dynamic payloads it is essentially seamless

There would be 4 additional functions/options:
1. `encrypt()`  Internal function, but allows users to manually encrypt data
2. `decrypt()` Internal function, but allows users to manually decrypt data
3. `setKeyIV()` Sets the Key and IV to use
4. `enableEncryption` Enables encryption in the background

As far as I can tell, it would work with the higher layers as well, but I haven't received a 2nd board to test with yet. So far I have been sending encrypted packets to another radio that is just repeating them back for decryption.

New Branch: https://github.com/TMRh20/nrf_to_nrf/tree/CCMEncryption

2bndy5, 2023-05-14:
This is pretty awesome! This is something I always wanted with the network layers to make them less transparent (and less susceptible to unauthorized nodes).

Looking at the code, I'm guessing that `CCM_ENCRYPTION_ENABLED` is only defined for the nRF52840. Seen as how only the nRF52840 (and the nRF5340) are the only chips to have this builtin, is it worth it to have this condition added to the `ifdef` gaurds?

I know there are external chips that people can use to do the same thing... Can these external chips be integrated on one node (say with RF24 lib) while another node uses nRF52850 and this lib's encryption features? I think the CRC checksum might need to be excluded from encryption for this idea.

TMRh20, 2023-05-14:
Hehe cool. 
Lots to think about. I'm not sure how exactly to handle the defines, maybe see if one of the CCM values are defined and then include the encryption stuff if so.

I would say the answer is yes, but this encryption is pretty specific, the datasheet says:
> CCM combines counter mode encryption and CBC-MAC authentication.

Since I was testing this with a RPi, I was wondering if it would be possible to decrypt the messages on the RPi, but don't know where to start. The same can be said for other chips, I would need a good reference like the NRF52840 datasheet.

It only seems designed to encrypt the radio data - not the CRC or radio addresses etc.

I still have questions about the NONCE generation:
> The NONCE vector (as specified by the Bluetooth Core Specification) will be generated by hardware based
on the information specified in the CCM data structure from CCM data structure overview on page 115 .

So I'm wondering if the counter value in the CCM Data structure needs to be set by the user or if that matters?

TMRh20, 2023-05-14:
Further testing shows problems with ACKs being sent too late. Maybe I would have to figure out how to on-the-fly encryption for this to work properly...

2bndy5, 2023-05-15:
> Further testing shows problems with ACKs being sent too late

I don't think the ACKs need encryption since ESB is a well known protocol (based on a some standard of wireless comms protocol/specs), especially since its (ab)used for reverse engineering. This should expedite the ACK sending, when ACKs don't have payloads. Its the outgoing payload data that is important to encrypt (eg RF24NetworkHeader info). Although, clearly I'm not as well versed on the builtin cryptographic co-processor.

> the payload size needs to be set 4 bytes larger than the payloads being sent/received

I hope this extra data isn't part of the decrypting necessities (see reconsideration below). Security-wise, it would be optimal if any info required for decryption is hard-coded into (or locally generated by) the app -- not acquired via external means. It might be problematic to expect users to compensate for an extra 4 bytes when setting the static payload size.

Looking at the datasheet, I might be thinking of the Cryptocell<sup>®️</sup> approach (chapter 6.6 in datasheet). The datasheet's description of Cryptocell<sup>®️</sup> tech seems to fit better with the various datasheets (that I skimmed a while ago) for other cryptographic ICs. CCM (a counter-based key generation mode) may be simpler if (or perhaps require) the counter is appended to the payload.

Ultimately, what I had dreamed of is way to implement an extra layer of validation after the RF24* libs did their native tasks. That way the crypto implementation is abstracted enough to not need code changes to the RF24Network layer.

---
Sorry if my understanding of this cyber-secure tech is woefully inadequate for this thread. Probably best to consider my input as estranged user feedback 😟.

2bndy5, 2023-05-15:
> I was wondering if it would be possible to decrypt the messages on the RPi, but don't know where to start. The same can be said for other chips, I would need a good reference like the NRF52840 datasheet.

You mean like [this Sparkfun ATECC508 breakout](https://www.sparkfun.com/products/15573),  or [this Sparkfun ATECC608 breakout](https://www.sparkfun.com/products/18077) (or [Adafruit's equivalent ATECC608 breakout](https://www.adafruit.com/product/4314)), or [Adafruit Infineon Trust M breakout](https://www.adafruit.com/product/4351)... There's also a plethora of chips that one could acquire [through mouser.com](https://www.mouser.com/c/semiconductors/integrated-circuits-ics/security-ics-authentication-ics/), but not all in that linked search are applicable here.

2bndy5, 2023-05-15:
I'm tagging @avamander for _any_ additional expertise. I hope this is not bothersome because I see he isn't subscribed to this repo's activity.

TMRh20, 2023-05-15:
>  This should expedite the ACK sending, when ACKs don't have payloads. Its the outgoing payload data that is important to encrypt

Even with just decrypting the incoming payload, with dynamic payloads it seems too big of a delay there to get the ACK back fast enough to the 24l01.

>I hope this extra data isn't part of the decrypting necessities

It is... here's where the 4 bytes come from: (page 111 of datasheet)
'The CCM block generates an encrypted keystream that is applied to input data using the XOR operation
and generates the **4 byte MIC field** in one operation.' ... 'The CCM terminology "**Message authentication
code (MAC)**" is called the "**Message integrity check (MIC)**" in Bluetooth terminology and also in this
document.'... ' **(CCM) mode is an authenticated encryption**
algorithm designed to provide both authentication and confidentiality during data transfer'

The MAC or MIC part is what I don't understand well, but this all together is supposed to do Authentication & Encryption.

TMRh20, 2023-05-15:
I'm not much of a BLE fan myself due to having trouble finding documentation and info.

I found that if I decrypt the data after sending an ack, it works. I think this is acceptable.

>Authentication may be overkill if the CRC validates, no?

I don't know, I just work here. Authentication should prevent replay attacks etc, but I'm not sure how its supposed to work.

2bndy5, 2023-05-15:
The documentation I can find ([here ya go](https://www.bluetooth.com/specifications/specs/)). Its finding the information that I'm looking for within those files that is the difficult part; you need to know _all_ the BLE terminology to even navigate the docs.

> Authentication should prevent replay attacks etc, but I'm not sure how its supposed to work.

I think the ESB's PacketID is a simplified way to prevent this, but it's size makes it harder to prevent replay attacks. It looks like the counter key is unique to the origin, which adds the info needed to identify replay attacks. IMO, this is yet another example of why I think the BLE protocol is so bloated. Unfortunately, the ESB's payload size is a tight constraint in comparison to BLE's massive payload size.

---
Obviously, my main focus is on the encryption. It would nice to have a way that could disallow neighbors from controlling someone else's home-automated devices that employ the RF24Network layer. There are other means of using the crypto peripheral: [adafruit/Adafruit_nRFCrypto](https://github.com/adafruit/Adafruit_nRFCrypto). I just never got around to trying it out.

2bndy5, 2023-05-15:
Did you have a specific use case in mind for this? Or are you just experimenting?

TMRh20, 2023-05-15:
Mostly just experimenting, but in the future it would be nice to have more secure networks. Right now I'm using the comm stack for sensors and switches (for LEDs etc), so its fairly simple. It would be nice to finish up the comm stack with something that makes up for lack of a presentation layer.

I've also found one or more of my neighbors was running exploits against my network (made some changes to RF24Network to fix), but nobody has managed to do anything significant in a while (that I am aware of) :p

TMRh20, 2023-05-20:
I've been playing around with this and included a randomized IV in each TX/RX using the built in Random Number Generator. From what I gather, you need to change the IV each transmission, so this implementation currently uses the full 8-byte IV + 4-byte MAC, so enabling encryption uses an extra 12-bytes. This leaves users with a 113-byte max payload size with 16-bit CRC enabled. Using static payloads, the payload size should typically be declared as `sizeof(payload) + 12;` the also read `sizeof(payload;` bytes.

2bndy5, 2023-05-22:
The [nRF24 Connect SDK crypto/aes_ccm example](https://github.com/nrfconnect/sdk-nrf/blob/main/samples/crypto/aes_ccm/src/main.c) isn't much help; it just generates the key, encrypts and decrypts data, then cleans up before exiting.

Just a friendly reminder, it would be nice to have any changes to the network layers pushed to a new branch. Their master branches are currently unusable for Arduino releases since the c'tors are defined differently for the nRF52840 boards.

Avamander, 2023-05-22:
>  Authentication may be overkill if the CRC validates, no?

No, not an overkill, because with such symmetric algorithms you can flip any chosen bit in the ciphertext and calculate a new passing CRC. 

Say you've got an encrypted message that contains one bit to control a lock, flip that bit in the ciphertext and it's flipped in the decrypted plaintext. So verifying the MIC is strongly recommended.

> I don't know, I just work here. Authentication should prevent replay attacks etc, but I'm not sure how its supposed to work.

If the IV contains a counter (and the integrity passes), you can ignore a counter value that is lower than what you've seen before.

> From what I gather, you need to change the IV each transmission, so this implementation currently uses the full 8-byte IV + 4-byte MAC, so enabling encryption uses an extra 12-bytes.

Yes, because reusing an IV leaks your key. 

With most block cipher modes (with counters) what is called the IV is often a concatenation of a nonce and a counter. In that case not incrementing the counter will compromise your key if you reuse a nonce (which you will, with the second packet). Incrementing it guarantees you won't be reusing an IV until counter overflow - you should be rekeying way before that point though.

TMRh20, 2023-05-22:
> Just a friendly reminder, it would be nice to have any changes to the network layers pushed to a new branch. Their master branches are currently unusable for Arduino releases since the c'tors are defined differently for the nRF52840 boards.

@2bndy5 Wut? I don't really understand what you mean by "unusable for Arduino releases..." or why.
In any case I'm done making changes to the higher level libs, I think I've got everything I need for the nrf_to_nrf driver in there, but I will create a new branch for any changes that do come along.

@Avamander Hey, long time no see! So are you saying I should be incrementing the packet counter in the CCM Data structure and sending it along with the IV in plaintext? I am generating a random IV each packet so should that make sure a nonce is not re-used? Also re-keying would be quite a challenge I think. Not sure how to go about doing that...

2bndy5, 2023-05-22:
>  I don't really understand what you mean by "unusable for Arduino releases..." or why.

Its related to nRF24/RF24Network#204. The current c'tor changes breaks RF24 lib usage on nRF52840 boards (which is why [that board is disabled in CI](https://github.com/nRF24/RF24Network/pull/202)). This is inconvenient when users would like to use nRF52 BLE + external nRF24L.

Avamander, 2023-05-22:
> Hey, long time no see! 

Hey!

> So are you saying I should be incrementing the packet counter in the CCM Data structure and sending it along with the IV in plaintext? 

Yep, the IV is not a secret (otherwise it would be a key).

> I am generating a random IV each packet so should that make sure a nonce is not re-used?

It might, it will be a problem after a certain amount of packets (the birthday attack). It's better to deterministically avoid that if possible.

With only 64bits for the IV you can only send 190,000 packets for the probability of a reuse to stay below one in a billion. 6,100,000 packets and it's one in a million chance for a reuse - one packet a second and you'll reach that point in around two months. Really really likely compromising your key.

> Also re-keying would be quite a challenge I think. Not sure how to go about doing that...

You can regenerate the nonce and restart the counter. Which should be fine for a while again, though keep in mind that nonce collisions will become more likely with each bit sacrificed for the counter.

2bndy5, 2023-05-22:
> the birthday attack

Some of these exploit names are rather funny 🤣 .

TMRh20, 2023-05-22:
@Avamander Err, adding the counter would be another 5-bytes, so 17-bytes of encryption overhead! This is getting a bit ridiculous lol, but if that's how it works, that's how it works I guess. So l guess I'll just reset the counter after every 200,000 packets or so? That seems easy enough.

@2bndy5 I can't think of a solution to that problem off the top of my head. Never really thought about using NRF24 with NRF52 devices + BLE. Could add a configurable define or something, but that adds to the current #ifdef soup.

Its great having people smarter than me to ask these questions! You guys rock lol!

Avamander, 2023-05-22:
> Err, adding the counter would be another 5-bytes, so 17-bytes of encryption overhead! This is getting a bit ridiculous lol, but if that's how it works, that's how it works I guess.

Well increasing the amount of bytes is one option, the tradeoff being how large the payload can be (but if you can segment your payload between multiple packets...). Though you could also just use 5 bytes of the IV for nonce and the rest 3 bytes for the counter, as you described.

TMRh20, 2023-05-22:
The datasheet describes this functionality as authentication + encryption, but I'm not sure how the authentication part is supposed to work. I can easily replay packets from my 'unauthenticated' NRF24L01 and the NRF52 as it stands will decrypt them and process the data. Any suggestions on how that's supposed to work with the MAC/MIC?

TMRh20, 2023-05-25:
So I've looked into it a bit and preventing replay attacks can be a bit of a challenge. If one had an RTC they could implement basic functionality that would only accept packets with a current timestamp. Otherwise, I guess one could go by the counter value, but it currently resets every 200,000 packets, so that may present some issues. This is quite interesting stuff imho.

TMRh20, 2023-05-26:
https://github.com/TMRh20/nrf_to_nrf/pull/9

closed 2023-05-26 by TMRh20