TMRh20/nrf_to_nrf #8: Issue with crashing w/RF24Network and higher libs w/Feather 82450 Express

TMRh20, 2023-05-20:
I just identified an issue with crashing, but it only seems to affect my new Feather 82450 Express board, not my XIAO BLE Sense board. In any case, it has to do with requesting a NULL value for pipe number in the `available()` function. On the Feather anyway, changing it to use a defined variable fixes the crashing. Logging this issue because it may also affect the RF24 library on different boards, it uses the same behaviour. Fix incoming.

closed 2023-05-20 by TMRh20

2bndy5, 2023-05-20:
Did you try using `null_ptr` instead?

2bndy5, 2023-05-20:
NM, I think use of `null_ptr` might need a `void*` parameter type.

TMRh20, 2023-05-21:
I just tested using `nullptr` and same behaviour. I think because its a null value, the device has nowhere to put the data, so it causes an overflow & subsequent crash. Took me forever to track this down too. Pretty sure it would be a good idea to change in RF24 as well.

2bndy5, 2023-05-21:
It is also good that you initialized the var to 0. This would be needed for consistent behavior in pyRF24 wrapper as well.

commit 0490d9bc (2023-05-20): Bug Fix affecting Feather 82450

- Use an actual variable instead of NULL when requesting pipe_num
closes #8