TMRh20/nrf_to_nrf #24: basic example does not compile on platformio

mkoenings, 2024-07-18:
Hello, 
I have used the nrf24 library successfully, on different hardware, thank you so much for this brilliant library!
Now I want to connect a stm32 with a nrf24 with a nrf52-DK,  with this library, but no luck..
Its the SDK too new maybe? Some member names are not valid.
Thanks!
 
the example does not compile on platformio
using the latest version of the nrf_to_nrf library with platform nrf52  version 10.5.0
build log:

`|-- nrf_to_nrf @ 1.2.11+sha.bf0c860

Building in release mode
Compiling .pio\build\nrf52_dk\libcee\nrf_to_nrf\nrf_to_nrf.cpp.o
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp: In constructor 'nrf_to_nrf::nrf_to_nrf()':
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:86:14: error: 'struct NRF_CCM_Type' has no member named 'MAXPACKETSIZE'   
     NRF_CCM->MAXPACKETSIZE = MAX_PACKET_SIZE;
              ^~~~~~~~~~~~~
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp: In member function 'bool nrf_to_nrf::write(void*, uint8_t, bool, bool)': 
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:455:24: error: 'struct NRF_RADIO_Type' has no member named 'EVENTS_TXREADY'; did you mean 'EVENTS_READY'?
             NRF_RADIO->EVENTS_TXREADY = 0;
                        ^~~~~~~~~~~~~~
                        EVENTS_READY
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:457:31: error: 'struct NRF_RADIO_Type' has no member named 'EVENTS_TXREADY'; did you mean 'EVENTS_READY'?
             while (NRF_RADIO->EVENTS_TXREADY == 0) {
                               ^~~~~~~~~~~~~~
                               EVENTS_READY
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:459:24: error: 'struct NRF_RADIO_Type' has no member named 'EVENTS_TXREADY'; did you mean 'EVENTS_READY'?
             NRF_RADIO->EVENTS_TXREADY = 0;
                        ^~~~~~~~~~~~~~
                        EVENTS_READY
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp: In member function 'bool nrf_to_nrf::startWrite(void*, uint8_t, bool, bool)':
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:647:20: error: 'struct NRF_RADIO_Type' has no member named 'EVENTS_TXREADY'; did you mean 'EVENTS_READY'?
         NRF_RADIO->EVENTS_TXREADY = 0;
                    ^~~~~~~~~~~~~~
                    EVENTS_READY
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:649:27: error: 'struct NRF_RADIO_Type' has no member named 'EVENTS_TXREADY'; did you mean 'EVENTS_READY'?
         while (NRF_RADIO->EVENTS_TXREADY == 0) {
                           ^~~~~~~~~~~~~~
                           EVENTS_READY
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:651:20: error: 'struct NRF_RADIO_Type' has no member named 'EVENTS_TXREADY'; did you mean 'EVENTS_READY'?
         NRF_RADIO->EVENTS_TXREADY = 0;
                    ^~~~~~~~~~~~~~
                    EVENTS_READY
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp: In member function 'void nrf_to_nrf::startListening(bool)':
.pio\libdeps\nrf52_dk\nrf_to_nrf\src\nrf_to_nrf.cpp:729:16: error: 'struct NRF_RADIO_Type' has no member named 'EVENTS_RXREADY'; did you mean 'EVENTS_READY'?
     NRF_RADIO->EVENTS_RXREADY = 0;
                ^~~~~~~~~~~~~~
                EVENTS_READY
*** [.pio\build\nrf52_dk\libcee\nrf_to_nrf\nrf_to_nrf.cpp.o] Error 1
`

2bndy5, 2024-07-18:
https://github.com/TMRh20/nrf_to_nrf/blob/bf0c86054f65bbe832be3e85ba2f0c2d5d3788a9/library.json#L16

Are you using the Arduino framework? 
Which [nRF52 DK](https://docs.platformio.org/en/latest/platforms/nordicnrf52.html#nordic) are you using?

This lib may be rather "new", but the nRF SDK certainly is not.

2bndy5, 2024-07-19:
I would first attempt to get it compiling in Arduino IDE (this is an Arduino library after all).

<details><summary>Notes from sandeepmistry Arduino core's README</summary>
<p>

There's an [extra step about flashing the softdevice to the board first](https://github.com/sandeepmistry/arduino-nRF5#selecting-a-softdevice). Despite the fact that this lib disallows BLE functionality, it still uses the softdevice API to access NRF_RADIO peripheral. There's also some [notes about the bootloader needing to be updated](https://github.com/sandeepmistry/arduino-nRF5?tab=readme-ov-file#nrf52).

</p>
</details> 

I was able to reproduce this in Arduino IDE (despite setting the softdevice in the tools menu) and PlatformIO.

<details><summary>Full compiler output (Arduino IDE)</summary>
<p>

```
FQBN: sandeepmistry:nRF5:nRF52DK:softdevice=s132
Using board 'nRF52DK' from platform in folder: C:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0
Using core 'nRF5' from platform in folder: C:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0

Detecting libraries used...
C:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10607 -DARDUINO_NRF52_DK -DARDUINO_ARCH_NRF5 -DNRF52 -DS132 -DNRF51_S132 -DNRF5 -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/gcc/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/device/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/drivers_nrf/delay/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/softdevice/s132/headers/ -DUSE_LFXO -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5 -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\variants\nRF52DK C:\Users\ytreh\AppData\Local\Temp\arduino\sketches\9C15DCBE3C71E0236B8913C72712F8B8\sketch\GettingStarted.ino.cpp -o nul
Alternatives for nrf_to_nrf.h: [nrf_to_nrf@1.2.11]
ResolveLibrary(nrf_to_nrf.h)
  -> candidates: [nrf_to_nrf@1.2.11]
C:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10607 -DARDUINO_NRF52_DK -DARDUINO_ARCH_NRF5 -DNRF52 -DS132 -DNRF51_S132 -DNRF5 -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/gcc/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/device/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/drivers_nrf/delay/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/softdevice/s132/headers/ -DUSE_LFXO -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5 -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\variants\nRF52DK -Ic:\Users\ytreh\Documents\Arduino\libraries\nrf_to_nrf\src C:\Users\ytreh\AppData\Local\Temp\arduino\sketches\9C15DCBE3C71E0236B8913C72712F8B8\sketch\GettingStarted.ino.cpp -o nul
C:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10607 -DARDUINO_NRF52_DK -DARDUINO_ARCH_NRF5 -DNRF52 -DS132 -DNRF51_S132 -DNRF5 -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/toolchain/gcc/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/device/ -IC:\Users\ytreh\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.8.0\cores\nRF5/SDK/components/drivers_nrf/delay/ -IC:\Users\y [... cut ...] d.org/platforms/Nordic-nRF52-DK) as the vendor URL in the [board's json config](https://github.com/platformio/platform-nordicnrf52/blob/1575f83394f16686c8df4516105b0ffb1e517a1c/boards/nrf52_dk.json#L49C11-L49C63).

---

> WARNING: library nrf_to_nrf claims to run on nordicnrf52, nrf52 architecture(s) and may be incompatible with your current board which runs on nRF5 architecture(s).

@TMRh20 I think this might be fixed by adjusting the library.properties. This was seen in the Arduino IDE output.

2bndy5, 2024-07-19:
Please add "on nRF52_DK" to this issue's title. We want specificity here, not a misleading generalization. This lib has been tested on various nRF52840 boards.

TMRh20, 2024-07-19:
It looks like this relates to compatibility issues between the NRF52840 and NRF52832.

For example, the registers described in the datasheet indicate there is no EVENTS_TXREADY register in the NRF52832. The same goes with CCM_MAXPACKETSIZE, it simply doesn't exist on the 52832.

In any case, the library would need some modifications to work with this device. Unfortunately without a 52832 device to test with, I'm not going to attempt to make the required changes.

mkoenings, 2024-07-19:
ok, fair point. I was hoping the  NRF52832 would be more or less more compatible with the ..40, but  that doesn't seem to be the case.  I will get a NRF52840 board and try again. Thanks for looking into this!

closed 2024-07-19 by TMRh20