TMRh20/nrf_to_nrf #26: network.update() and network.available() handling

flymanster, 2025-01-06:
Hello,

is it possible to handle network.update() and network.available() by interrupt routine instead of main loop?

BR,
fly_man

2bndy5, 2025-01-06:
You should be able to, but understand that `radio.available()` is not the same as `network.available()`; you still need to call `network.update()` and `network.available()` when an interrupt occurs.

Just hook up your radio's IRQ pin and listen to when it falls LOW (HIGH means inactive). Use [`RF24::maskIRQ()`](https://nrf24.github.io/RF24/classRF24.html#abf68b9b0c9cd17179e9e144c3e7f9c45) to only let RX events trigger the IRQ pin...
```cpp
volatile bool isAvailable = false;

#define IRQ_PIN 2

void handler() {
  bool tx_ds, tx_df, rx_dr; // declare status flags for IRQ event type
  radio.whatHappened(tx_ds, tx_df, rx_dr); // clear interrupt and get status flags
  // now, only `rx_dr` should be `true` because we called `maskIRQ()` in `setup()`
  isAvailable = rx_dr;
}

void setup() {
  // ...
  pinMode(IRQ_PIN, INPUT);
  attachInterrupt(IRQ_PIN, handler);
  radio.maskIRQ(true, true, false); // only trigger when data is received
}

void loop() {
  if (isAvailable) { // radio has data in the RX FIFO
    network.update(); // handle data in radio's RX FIFO
    while (network.available()) { // handle network data queued from the RX FIFO
      // do stuff with incoming network data
    }
    isAvailable = false; // reset the flag for next IRQ event
  }
  // ...
}
```

flymanster, 2025-01-06:
Thank you for quick response and I'm sorry not sufficient description. I'm running this lib with success on nRF52840 and so looking solution with internal interrupts handling.

2bndy5, 2025-01-06:
With the nRF24L01 (using RF24 lib), that's closest offering to any kind of interrupt handling. If you're using [nrf_to_nrf lib](https://github.com/TMRh20/nrf_to_nrf) on nRF52840, then there is no interrupt handling (yet).

By putting IRQ handling in the responsibility of the user, we (lib maintainers) don't have to worry about race conditions (concerning the SPI bus activity), dangling pointers, and all the complexity with multiprocessing (RTOS vs bare-metal).

2bndy5, 2025-01-06:
Oops. I didn't realize this issue was for nrf_to_nrf repo. I thought it was for RF24Network.