TMRh20/nrf_to_nrf #27: communication between two nRF52840 devices

lon2347, 2025-01-19:
Hello ,

Thanks for your library.  

Could I use the library between two nRF52840 which one is transmitting data via ESB and another use the library to receive data ?

Thanks.

TMRh20, 2025-01-19:
Yup, just use any of the included examples. Please see the documentation to utilize the full capabilities like large payload sizes etc as the radios are configured to work with nRF24 out of the box

lon2347, 2025-01-19:
Thanks for your quickly reply!

what about nRF52810? when I compile library using nRF52810 Board in Arduino IDE it throw error said there is no member in some structs defined in related h file. I knew the nrf_to _nrf is for providing communication between nRF52840 and nR24L, however not sure if it support nRF52810.

Thanks.

TMRh20, 2025-01-19:
Currently it looks like there are some registers that are different between the two types of devices, it can probably be worked around, but for now, no the library is not compatible with 52810 devices. 
This is something I will note for future reference, but unless I get ahold of a 52810 device myself, I will not likely add support for these boards.

closed 2025-02-06 by TMRh20