TMRh20/nrf_to_nrf #28: Periodic radio power-up/shutdown for low battery consumption

petermeter69, 2025-04-09:
Hello,
I want to use this great library for a battery operated device where current consumption is crucial. 
The device does some sensor data acquisition/transmission (ca. 1 second), then it sleeps for 29 seconds (thus sensor cycle is 30 seconds). The battery consumption in sleep state is of primary importance for maximizing battery life. 
I am aware of different low power modes for NRF52, like:

- System off, external interrupt wake-up, current draw ca. 1 uA
- System on, flash off, in- or external interrupt wake-up, ca. 3-4 uA 
- System on, flash on, in delay(), ca. 25 uA

Since my app needs context variables across cycles and I do not have an external interrupt source, I am fine with System on flash on state and with  its ca. 25 uA current draw. 
With a fresh Arduino sketch (no Serial, no extra consumers), I can achieve it in a delay while measuring battery current with a  multimeter:
`digitalWrite(LED_GREEN, LOW); //LED ON`
`delay(3000);                  //130 uA (LED + Flash)`
`digitalWrite(LED_GREEN, HIGH);//LED OFF`
`delay(3000);                  //24 uA (Flash only)`

If I add the library's #include statements and variable declarations, without further initializing radio, and network variables the current draw increases to ca. 700 uA.  
`#include "nrf_to_nrf.h"`
`#include "RF24Network.h"`

`nrf_to_nrf radio;`
`RF52Network network(radio);`
`//no other radio or network call`
...
`digitalWrite(LED_GREEN, LOW); //LED ON`
`delay(3000);                  //806 uA (radio, LED, Flash)`
`digitalWrite(LED_GREEN, HIGH);//LED OFF`
`delay(3000);                  //692 uA (radio, Flash only)`

On initializing radio, battery current increases to 11.47 mA (without transmission):
`radio.begin()`
`radio.setChannel(125);`
`radio.setPALevel(0);`
`radio.setDataRate(NRF_1MBPS);`
...
`delay(3000);                  //11.470 mA (initialized radio, Flash)`

The problem arises on shutting down radio, when some major consumer remains powered with almost 1mA of current draw:
`digitalWrite(LED_GREEN, LOW);  //LED ON`
`InitRadio()`
`delay(3000);                                //11.470 mA (initialized radio, LED, Flash)`
`digitalWrite(LED_GREEN, HIGH);//LED OFF`
`radio.powerDown();`
`delay(3000);                               //926 uA (radio after Shutdown, Flash)`

I tried calling sd_power_system_off();  instead of radio.powerDown() with same results. 
Of course, a current draw of ca. 1mA is anything but ideal for sleep mode. 

Question: what do you think causes this high current draw after radio shut down, and how could it be eliminated? 

Thanks 
Peter

TMRh20, 2025-04-09:
I'm not sure on this one. I'm tempted to say its a hardware & programming issue, not really a library issue, because this lib just handles the radio and required peripherals, it doesn't have anything to do with power management.

However, the radio relies on the HF and LF clocks, so maybe disabling those will make a difference?
```
NRF_CLOCK->TASKS_HFCLKSTOP = 1;
NRF_CLOCK->TASKS_LFCLKSTOP = 1;
```

Other than that I don't know what to suggest at this point.
Please let me know how it goes. I would have to do further testing to see what is going on.

petermeter69, 2025-04-10:
First, thanks for your reply! 
With the library _#includes_ and declared _radio/network_ variables, an empty _setup()_ function, and below _loop()_ function, program execution stops while disabling LFCLK and RED/GREEN is blinked only once.
` digitalWrite(LED_RED, LOW);                   //
  delay(3000);                                             //1.34mA

  digitalWrite(LED_RED, HIGH);                  //
  delay(3000);                                             //0.69mA

  InitRadio();
  digitalWrite(LED_GREEN, LOW);              //
  delay(3000);                                            //11.4mA

  radio.powerDown();
  NRF_CLOCK->TASKS_HFCLKSTOP = 1; //
  NRF_CLOCK->TASKS_LFCLKSTOP = 1;  //Execution stops here 
  digitalWrite(LED_GREEN, HIGH);           //
  delay(3000);                                          //0.69mA`

By commenting out  NRF_CLOCK->TASKS_LFCLKSTOP = 1; the loop executes infinitely, and the current draws are unchanged. I suspect Arduino delay() must be derived from LFCLK. Interestingly enough, when execution stops on LFCLK STOP, the green led succeeds to be unlit, i.e. the LFCLK does not stop immediately.   

Disabling HFCLK seems to be a step in the right direction (0.92mA vs. 0.69mA), but we are not there yet...

TMRh20, 2025-04-10:
I don't have a way to properly test this right now, beyond a bench power supply that only measures down to 1mA, and it ends up measuring 0 when I run this:

```
#include <nrf_to_nrf.h>
#include <RF24Network.h>

nrf_to_nrf radio;
RF52Network network(radio);

void setup() {
  
  radio.begin();
  network.begin(00);
  delay(5000);
  radio.powerDown();

}

void loop() {

  delay(1000);

}
```

So if you run the above code what do you get?
Can you post a complete example of your code?

petermeter69, 2025-04-10:
Running your code yields the same results: 11.4mA in the first 5secs, then 0.9mA. You will only be able to measure uA currents with a 20mA DC range or smaller 4digit or better multimeter. 

I develop for XIAO NRF52840 with the Seeed 1.1.9 Boards support package. I published all code in chunks. The LEDs are added to have some visual feedback of what is going on. But let us stick with your above code base for future. 

I also have some old Adafruit Feather and ItsyBitsy NRF52840 boards as well.  With the ItsyBitsy (it also has a battery pin) and the Adafruit NRF52 1.6.1 BSP, I measure similar current draws for your above code: 12.6mA in the first 5secs, and then 1.870mA. 

The reason  I started with a blank Arduino sketch without radio is to have a proof that my particular Xiao NRF52 board is capable of 25uA of current in delay() without Serial and Radio. On top of that I only added your above code (#includes, declarations, and InitRadio) and the sleep current after radio.powerDown() jumped to around 1mA. I conclude from that that the radio initialization must have caused it. Maybe I am wrong.

petermeter69, 2025-04-10:
This is how I connect to the Battery pin:
[image]

Below is the state when 23.4uA can be measured in delay():  
[image]

TMRh20, 2025-04-10:
> I conclude from that that the radio initialization must have caused it.

I'm getting the same idea.

The only solution I can think of is to get myself a capable meter & test a bit, but that will take some time.
I haven't played around with the lower-power modes of the 52840s yet, so am not very knowledgeable in that regard. In the meantime you could also try the XIAO forums, there are others that know much more about this stuff than me.

petermeter69, 2025-04-13:
Sunny Sunday here, best time to investigate excessive current consumption.
Below code shows that it is the constructor of the radio variable alone that raises current draw. This state persists after the variable goes out of scope.   

`delay(3000);            //24uA`
`{`
`    nrf_to_nrf radio;`
`    //RF52Network network(radio);`
`    delay(3000);        //700uA`
`}`
`delay(3000);           //700uA, but should be 24uA`

Further to variable initializations, the constructor of nrf_to_nrf contains several hardware calls:

`nrf_to_nrf::nrf_to_nrf()`
`{`
`...`
`    setAutoAck(1);`
`...`
`#if defined CCM_ENCRYPTION_ENABLED`
`    NRF_CCM->ENABLE = 2;`
`...`
`    NRF_RNG->TASKS_START = 1;`
`#endif`
`};`

setAutoAck calls disableDynamicPayloads(), which sets NRF_RADIO->PCNF0 and PCNF1 before NRF Radio is enabled in begin(). 
Since CCM_ENCRYPTION_ENABLED is defined, CCM and RNG are also enabled/started. Both cause unnecessary use of power (at least from the point of view of a battery operated device with no encryption).

By commenting out setAutoAck/CCM/RNG in the constructor, current draw normalizes (even with network uncommented):
`delay(3000);            //24uA`
`{`
`    nrf_to_nrf radio;`
`    RF52Network network(radio);`
`    delay(3000);        //24uA`
`}`
`delay(3000);           //24uA`

With radio initialization/shutdown, however, there remains a problem:
`radio.begin();`
`network.begin(00);`
`delay(3000);           //11.4mA`
`radio.powerDown();`
`delay(3000);          //0.22mA`

This 0.22mA is due to HFCLK remaining powered. By switching it off, too, current draw drops to the anticipated level:
`radio.begin();`
`network.begin(00);`
`delay(3000);           //11.4mA`
`radio.powerDown();`
`NRF_CLOCK->TASKS_HFCLKSTOP = 1;`
`delay(3000);          //24uA`

Recommendations for running this library on battery operated devices:

1. Remove hardware calls from the constructor. 
2. Do not initialize CCM/RNG if encryption is unneeded.   
3. Stop HFCLK in radio.powerDown()

The true strength of NRF52 lies in its low power consumption. It draws at least a magnitude lower current than comparable ESP devices. The nrf_to_nrf is a truly fantastic library for communicating with NRF52/NRF24 devices. On implementing these modifications, it will also be best choice for battery operated devices.

TMRh20, 2025-04-13:
I think this is all do-able, it will probably just take me a bit of time to make the adjustments, but I really appreciate the great investigation and feedback!

I will probably get a chance to get started on this over the next week or two.

2bndy5, 2025-04-13:
Maybe keep the issue open until it is resolved in the lib code.

@petermeter69 Code blocks in MarkDown can be enclosed in triple backticks:
~~~
```cpp
delay(3000);            //24uA
{
    nrf_to_nrf radio;
    RF52Network network(radio);
    delay(3000);        //24uA
}
delay(3000);           //24uA
```
~~~

Notice the `cpp` part enables syntax highlighting for C++ code blocks. The above will then render as:
```cpp
delay(3000);            //24uA
{
    nrf_to_nrf radio;
    RF52Network network(radio);
    delay(3000);        //24uA
}
delay(3000);           //24uA
```

TMRh20, 2025-04-14:
@petermeter69 I just created a pull request from the PowerManagement branch of this repo. 

1.  Removed hardware calls from the constructor.
2.  Do not initialize CCM/RNG if encryption is unneeded.
3.  Stop/start HFCLK in radio.powerDown()/radio.powerUp()
4.  Stop/start RNG in radio.powerDown()/radio.powerUp() as required by encryption

If you get a chance in the next little while to test it out, please let me know if it addresses all the issues.

TMRh20, 2025-04-14:
@petermeter69 You don't have to do anything, I had it set to close the issue automatically when the code was pushed. GitHub is cool like that.

PR #29 (2025-04-14): Power Management Changes per #28
- Configure AA in begin() function instead of c'tor.
- Move CCM configuration to the setKey() function instead of c'tor
- Move code block to stopListening from write() and startWrite() functions so its only called from one place
- Start/stop the HF Clock on powerUp/powerDown
- Start/stop the RNG on powerUp/powerDown

Closes #28
closed 2025-04-14 by TMRh20